free-pdk/easy-pdk-programmer-hardware #5: [Question] Support for PMC232 PMC234 PGC433 PGC434?

DanielMazurkiewicz, 2019-08-27:
Do you plan by any chance do add support in programmer for these microcontrollers?

PMC232
PMC234
PGC433
PGC434

spth, 2019-08-29:
The PMC232,  was officially released 3 days ago (I don't know about actual availability). The PGC433 and PGC434 are still considered "under development". The PMC234 datasheet was released 4 days ago.
While support for these will be very useful, I guess it will take some time.

freepdk, 2019-08-30:
Support can be added only, if a real physical device is available. It is not possible to add support for data sheet only CPUs.

I do have PMC232 and PMC234. Do you have a project and want to use them? If so I will put them on top of the priority list for device support.

DanielMazurkiewicz, 2019-09-01:
>  If so I will put them on top of the priority list for device support.

@freepdk Thank you, that is a very kind of you, but no rush with that. Would like to at some point drive brushed DC motors and having these industrial grade microcontrollers (PFC and PMC series) supported will probably be a must for such a projects. And after couple of days of research among different brands my priority wish list (based on price and capabilities I will need) would look more like that:

1. PMC156 - SOP16
2. PMC150 - SOP8
3. PMC251 - SOP14

Analyzed that in some cases it will be easier to program and cheaper to have two or even three of above plus some external comparator (if necessary) than having one uC that provide all functions on board.

freepdk, 2019-09-02:
@DanielMazurkiewicz 

PMx150C is well supported already
PMx156 I can add next
PMC251 is partially supported already (probe/read works, write I need to finish)

DanielMazurkiewicz, 2019-09-02:
Used that file: 
https://github.com/free-pdk/easy-pdk-programmer-software/blob/master/fpdkicdata.c
as reference for table below:

| Microcontroller        | Programmer support | SDCC support |
|:----------------------:|:------------------:|:------------:|
|      PFS154            |         ✓          |              |
|      PFS172            |                    |              |
|      PFS173            |         ✓          |              |
|      PMS130            |                    |              |
|      PMS131            |                    |              |
|      PMS132B           |                    |              |
|      PMS133            |                    |              |
|      PMS134            |                    |              |
|      PMS150            |                    |              |
|      PMS150C           |         ✓          |              |
|      PMS150G           |                    |              |
|      PMS152            |                    |              |
|      PMS153            |                    |              |
|      PMS154B           |         ✓          |              |
|      PMS154C           |         ✓          |              |
|      PMS156            |                    |              |
|      PMS171B           |                    |              |
|      PMS232            |                    |              |
|      PMS234            |                    |              |
|      PMS271            |                    |              |
|      PFC151            |                    |              |
|      PFC154            |                    |              |
|      PFC232            |                    |              |
|      PFC433            |                    |              |
|      PFC434            |                    |              |
|      PMC131            |                    |              |
|      PMC150            |         ✓          |              |
|      PMC153            |                    |              |
|      PMC156            |                    |              |
|      PMC232            |                    |              |
|      PMC234            |                    |              |
|      PMC251            |     partial        |              |
|      PMC271            |                    |              |

Feel free to do whatever you want with that table (would be nice to have it somewhere in readme). Here is a markdown for it:
```
| Microcontroller        | Programmer support | SDCC support |
|:----------------------:|:------------------:|:------------:|
|      PFS154            |         ✓          |              |
|      PFS172            |                    |              |
|      PFS173            |         ✓          |              |
|      PMS130            |                    |              |
|      PMS131            |                    |              |
|      PMS132B           |                    |              |
|      PMS133            |                    |              |
|      PMS134            |                    |              |
|      PMS150            |                    |              |
|      PMS150C           |         ✓          |              |
|      PMS150G           |                    |              |
|      PMS152            |                    |              |
|      PMS153            |                    |              |
|      PMS154B           |         ✓          |              |
|      PMS154C           |         ✓          |              |
|      PMS156            |                    |              |
|      PMS171B           |                    |              |
|      PMS232            |                    |              |
|      PMS234            |                    |              |
|      PMS271            |                    |              |
|      PFC151            |                    |              |
|      PFC154            |                    |              |
|      PFC232            |                    |              |
|      PFC433            |                    |              |
|      PFC434            |                    |              |
|      PMC131            |                    |              |
|      PMC150            |         ✓          |              |
|      PMC153            |                    |              |
|      PMC156            |                    |              |
|      PMC232            |                    |              |
|      PMC234            |                    |              |
|      PMC251            |     partial        |              |
|      PMC271            |                    |              |

```

freepdk, 2019-09-02:
Hi,

in development branch are far more supported ICs (most of them Probe/Read only for now).

"easypdkprog list" shows all supported ICs:

$ ./easypdkprog list
Supported ICs:
 PMC251   (0x058): OTP  : 1024 (16 bit), RAM:  96 bytes
 PMS132   (0x109): OTP  : 2048 (14 bit), RAM: 128 bytes
 PMS132B  (0x109): OTP  : 2048 (14 bit), RAM: 128 bytes
 PMS150C  (0xA16): OTP  : 1024 (13 bit), RAM:  64 bytes
 PMS152   (0xA27): OTP  : 1280 (14 bit), RAM:  80 bytes
 PMS271   (0xA58): OTP  : 1024 (16 bit), RAM:  64 bytes
 PFS154   (0xAA1): FLASH: 2048 (14 bit), RAM: 128 bytes
 PMS133   (0xC19): OTP  : 4096 (15 bit), RAM: 256 bytes
 PMS134   (0xC19): OTP  : 4096 (15 bit), RAM: 256 bytes
 PMS131   (0xC83): OTP  : 1536 (14 bit), RAM:  88 bytes
 PMS171B  (0xD36): OTP  : 1536 (14 bit), RAM:  96 bytes
 PMS154B  (0xE06): OTP  : 2048 (14 bit), RAM: 128 bytes
 PMS154C  (0xE06): OTP  : 2048 (14 bit), RAM: 128 bytes
 PFS173   (0xEA2): FLASH: 3072 (15 bit), RAM: 256 bytes

As soon as Write support is added and tested I want to merge this to master version and make a real release.

spth, 2019-09-03:
SDCC, while not including headers yet, can compile programs for them all, except the PMC251 and PMS271.

Philipp

DanielMazurkiewicz, 2019-09-03:
> "easypdkprog list" shows all supported ICs
@freepdk it requires installing easypdkprog to get to know what is supported and what not, so I still think it would be nice to have it in a documentation

freepdk, 2020-07-17:
The website https://free-pdk.github.io/ is now having a list of truly supported ICs.

closed 2020-07-17 by freepdk