free-pdk/easy-pdk-programmer-hardware #7: VPP/VDD voltage capped to 3.45V

cirlam, 2019-10-02:
Hi, 

Firstly, thank you for all the hardwork developing this programmer!

I have a similar (ish) problem to [https://github.com/free-pdk/easy-pdk-programmer-hardware/issues/6](https://github.com/free-pdk/easy-pdk-programmer-hardware/issues/6)  with a hand soldered pcb.

When using easypdkprog probe, It responds with no MCU found (PFS-154C in the socket). 

Using easypdkprogtest, it only gets to 3.46v for vdd and 3.44v for vpp. I've confirmed these values with a multimeter. Voltage around the diodes is identical to Paul - (-2.7v D3<->GND, and 0.0v D2<->GND, -1.5v<->GND for both diodes on the shared side). 

If I change easypdkprogtest to request 1v or 2v it works perfectly, however it cannot go any higher than this. If I change easypdkprogtest to 100.0, it remains at 3.46v/3.44v. 

Visual check of the PCB with microscope looks OK from what I can see.

Any ideas of what I should be checking to resolve this? Thank you!

freepdk, 2019-10-04:
Hi,

this sounds like the dc dc is not generating the 14V. Look at the coil / big diode / the DC DC ic and measure voltages there (when easypdkprogtest is running).

==> check U6  when easypdkprogtest is running:
pin 4/5: input voltage (should be 5V, if is not 5V check the 2 transistors Q1 and Q2)
pin 1: output of high voltage going to big diode (should be around 14V)
pin 3: input of feedback voltage (should be something >5V, sorry can't measure right now) 

My guess: big diode is soldered backwards or Q1/Q2 is mixed up.

cirlamOB, 2019-10-04:
Thank you for the help! I will check these this evening and get back to you.

cirlam, 2019-10-04:
Thank you for the help!

I will check these this evening and get back to you.

cirlam, 2019-10-05:
Success! I mistakenly soldered a 100k resistor to R1. Now works fine and detects IC's ok.

freepdk, 2019-10-05:
Great to hear.

Wrong R1 / R5 values will cause a wrong feedback to the DC DC so the output voltage will be incorrect.

Thanks for the feedback.

closed 2019-10-05 by freepdk