free-pdk/easy-pdk-programmer-hardware #11: Medium-run buy?

cnlohr, 2019-12-20:
I'm toying with spending some $$ and doing a run of about QTY 200 or so of these from somewhere like Elecrow (and letting them sell on their site).  I've done a lot of medium run projects with various Chinese houses.  

Though the hardware is definitely not the way I would have designed it, it seems you guys are pretty happy with it.  I am happy to deal with any production issues and get a bunch of these made and available for sale to anyone.  

The question comes - is anyone else already doing this?  Would you guys modify some aspect of the design? Are there changes I should be making for production?  Would you want to roll them back?

freepdk, 2020-01-01:
Hi,

right now there are no known issues. Almost everything of the designed worked out perfectly.
But not all processors are added at the moment so maybe some changes might be needed in future.

Based on the response and feedback, I don't think there is a demand for 200 pcs.

For serious projects you always can use the (not so expensive) original programmer which is available here: https://lcsc.com/product-detail/Emulator_PADAUK-Tech-5S-P-003_C263009.html
(even if it is listed out of stock, just do an order and usually LCSC sources it within 1 or 2 weeks).

cnlohr, 2020-01-14:
I am much more interested in the open source stack - and your processes are much cleaner for makefile-based implementations.  

I think if I actually do the buy and some how-to videos the demand would expand significantly.  Do you see any issue with me proceeding otherwise?  I.e. I just want your guys blessing before I go get elecrow spun up and ask them to sell your product in their shop.

freepdk, 2020-01-15:
The license for this hardware project is "CC-BY-SA-4.0" which in short gives the following permissions: 

- Commercial use
- Modification
- Distribution
- Private use

Under the following simple conditions:

- License and copyright notice
- State changes
- Same license 

Details here: https://github.com/free-pdk/easy-pdk-programmer-hardware/blob/master/LICENSE

As long as you reference this original project (e.g. URL to this repository or to free-pdk.github.io) you are free to do whatever you want.

closed 2020-07-17 by freepdk