free-pdk/easy-pdk-programmer-hardware #15: Should the design be easy to get via JLCPCB (and distribution in the EU be feasible)?

spth, 2020-05-25:
AFAIK, the design uses a lot of components from LCSC. LCSC cooperates with JLCPCB for board making and assembly.
While JLCPCB is willing to add some LCSC parts on request, however JLCPCB explicitly does not support connectors (they say connectors interfere with their approach of doing assembly before depanelization).

This makes the following option attractive:

* Have the connectors as through-hole components to make them relatively easy to hand-solder.
* Make the rest of the design orderable at JLCPCB

This has the added advantage of making it possible to sell the device as a kit in Europe: In the EU, it is practically illegal for private individuals and small companies to make electronic devices to sell or give away fro free (WEEE Directive). However, AFAIK, it is still legal to sell kits that require substantial assembly, such as soldering.

At first sight, it seems the only change required would be to use a through-hole USB connector.

cpldcpu, 2020-05-25:
I like the idea of the trough hole usb connector. Jlcpb is not able to
mount any usb connector so some solution is needed anyways. It would be a
huge usb-a Plug though. No usb-c socket available in that

cpldcpu, 2020-05-25:
Another component that is bizarrely missing from jlcpcbs catalogue is a proper power inductor, btw.

spth, 2020-05-25:
I have been in contact with JLCPCB, and they are willing to add LCSC surface-mount parts on request to their catalogue (long term goal is to have them all anyway), with the exception of connectors.

cpldcpu, 2020-05-25:
Yes, but most components will stay as "extended", which means there is an additional set up fee. It makes a lot of sense to redesign to only use "basic" components. So far that works, the only extended component is the MCU.

spth, 2020-05-25:
I guess a set up fee is not a big deal as long as it doesn't affect too many components. Do you know how much the setup fee per part is.

cpldcpu, 2020-05-25:
It's $3.00. Does not sound much, but consider that they will assemble a maximum of 5 boards. So each board will be $0.60 more expensive for every non-basic part that is used.

cpldcpu, 2020-05-25:
But it's not an issue, since the "lite" design (see EEV forum) is only based on "basic" components.

spth, 2020-05-25:
> It's $3.00. Does not sound much, but consider that they will assemble a maximum of 5 boards. So each board will be $0.60 more expensive for every non-basic part that is used.

I just checked their website. It seems the limit has been raised to 50 boards. And for those wanting to go over the limit, they recommend panelizing the design (I guess they would then assemble 50 panels). Seems enough for now.

cpldcpu, 2020-05-25:
Well, I frequently use their service and two weeks ago it was still limited to 5.
Panelizing is something that would help, but easyeda does not support panelizing of the component placement. It's rather weird really. A workaround would be going to the KiCAD design.

vanmierlo, 2020-05-29:
When the connectors are left out, there is exactly one through-hole component left: the crystal. Would it not be prudent to replace that by an SMD variant then?

cpldcpu, 2020-05-31:
There are also the header for the programming, which is through hole.

The crystal is optional anyways, since the MCU is able to synchronize from USB.

freepdk, 2020-06-01:
@cpldcpu

In case this needs to be identified as a board variant (which I think might not be required) it would be a good idea to connect one of the unused STM32 pins (e.g. choose one from PB8-PB15) to GND.
This would make it possible to identify this hardware variant in the firmware and to handle things different in case it's needed.

cpldcpu, 2020-06-01:
>In case this needs to be identified as a board variant

I conntected PB8 to GND in the design. I posted a changelog in the EEVblog thread.

kasbah, 2020-10-30:
I'd be interested to help organise a group-buy of a JLC assembled PCB through [kitspace.org](https://kitspace.org). We have just begun testing the waters of a group-buying functionality (https://github.com/kitspace/kitspace-v2/pull/13). 

We deployed a basic shop, which, for now, is just has a straight-forward ["buy it now" of components for my friend's workshop](https://shop.kitspace.org/buy/electron-detector-parts-only/). But we are able to take payment, take care of shipping everything out internationally, etc. 

If we can gather all the required info together, agree on how any profit needs to be shared, etc. I can set up a page where people can register interest and pay and organise the manufacture and shipping out (or issue full refunds if MOQs are not met).

kasbah, 2020-11-01:
Hi @cpldcpu, great, I didn't realise you were the same person as tim_ on EEVblog. I'll email you. 

The idea is that Kitspace could take care of everything that needs to be done to get the assembled boards into the hands of the people that want them. So that includes ordering, shipping, customs, handling refunds etc. Of course we want to make sure people that did the design work get a fair share of the money too. 

Regarding the MCU stock, I've actually been coding up a stock-alert web app for JLC that emails you when your desired quantity is back in stock. Let's see if I can get that finished so it can help with this.

cpldcpu, 2020-11-01:
hi @kasbah, indeed, waiting for your mail then.

Regarding sharing of any leftover money, I guess we'd need to figure out something that helps the community. I certainly would not like to take any direct renumeration. The programmer was developed by @freepdk, I just made some tweaks to it to make the design more compatible to JLCPCB and did the test runs and evaluation.

Priority should be to make the programmer more accessible, as this has been a difficulty for many.

I don't know whether anybody else has opinions this?

cpldcpu, 2020-11-01:
Btw, here is the current state of the lite programmer design log. That's the best source for information I have right now.

[EasyPDKprog_lite.pdf](https://github.com/free-pdk/easy-pdk-programmer-hardware/files/5470996/EasyPDKprog_lite.pdf)

cpldcpu, 2020-11-02:
@kasbah 

I elected to publish it to a separate repository instead of an add on to this one, to avoid too much confusion. You can find the lite version here: https://github.com/free-pdk/easy-pdk-programmer-lite-hardware

It's still incomplete. @freepdk can you check whether everything is ok? I noticed you changed the license recently, will update to CC.

cpldcpu, 2020-11-25:
@kasbah 

Just checking: is this useful to you in any way? Are there still open questiosn? (Not sure how to translate the eyes emoji ;) )

kasbah, 2020-11-25:
Hi, yes very useful indeed, thanks. By "eyes" I just meant, at the time: I've seen this and will try it out later, ha.

Later I went through the process of uploading it to JLC assembly, which seemed to work fine.  But as you commented before, crucial components are out of stock at the moment so it wasn't enough to get a cost estimate. (I need to get my stock-alert thingy finished.) I also noticed one component which we could maybe switch out for something that JLC normally carries, will take another look to give you an exact part.

cpldcpu, 2020-11-25:
Ah ok, thanks for feedback. Someone in another forum mentioned that many of the STM microcontrollers are on allocation right now. So that may not be a problem limited to JLCPCB. Hopefully we will see new stock soon.

EDU-BR-TAR, 2021-06-03:
I have a humble idea to lower the cost of the board, I can make a shield for the bluepill board many people have this board in their hands, I just need someone's help to modify the firmware for the stm32f103c8t6

freepdk, 2021-06-03:
> I have a humble idea to lower the cost of the board, I can make a shield for the bluepill board many people have this board in their hands, I just need someone's help to modify the firmware for the stm32f103c8t6

There was a discussion about this already. STM32F10x does not have 2 DACs which are needed in order to provide the reference voltages for the opamp.

**And a big heads up to all people thinking they have a so called "BluePill" board.** 
In some other projects I did, it turned out that >90% of all sold bluepill boards come with a FAKE STM32 IC. Some of them (using the GD32F103) seem to be 95% compatible but some using the cheaper fakes (CS32F103/CH32F103) struggle with many problems (even USB is not working without adding quirks to the firmware). The most annoying fact is that the fake ICs  also come with a fake label claiming to be a "real" STM32F103. Dealing with this situation in a hobby open source project is something I can strongly recommend to AVOID.

My point of view: 
==> There is no STM32F103 anymore which "many people" have (almost all of them buying clones)
==> Do not use STM32F103 for any project where you do not manufacture the complete hardware yourself and you source the IC directly from ST / DigiKey / Mouser / ...
==> Do not create add on hardware for devices which most likely contain clone STM32F103. You will waste more time dealing with the clone problems than you can spent on the actual project.

EDU-BR-TAR, 2021-06-04:
i understand , thanks for answering