free-pdk/easy-pdk-programmer-hardware #16: No easy way to thank you, short of opening an issue

brainsmoke, 2020-08-27:
I ordered and assembled your programmer and it works perfectly!
But not I'm having difficulty to find a method of thanking you, short of opening an issue.

Once this whole pandemic is over I would like to buy you a beverage at some hacker con
if you're into visiting those.

Cheers,
Erik

[image]

sengiv, 2021-06-08:
how much did this cost you?
where did you source the parts?

looking to build one like yours, but my cost (lcsc.com) seems to high.

brainsmoke, 2021-06-08:
The STM32F072 has been going up in price from US$2.50 to US$6.92 (and it's out-of-stock)
I paid US$23.65 for enough parts for three (with extras for many parts), but the same parts list
is US$33 now excluding the STM32.

This is the BOM for the order for comparison: (I might have left out stuff that I had already)
At the time the boost converter was not in stock, so I used a compatible part.

[LCSC_Exported_20210609_011821.csv](https://github.com/free-pdk/easy-pdk-programmer-hardware/files/6618480/LCSC_Exported_20210609_011821.csv)

sengiv, 2021-06-09:
@brainsmoke thanks for your reply, you're right prices have changed. At the current prices it's becoming more economical to get the original programmer : )

Btw, if you dont mind sharing some details about your project. What are you using this mcu's for? Production or hobby? Would you say these mcu's are worth the extra work? 

As far as I can see the extra cost & work doesnt make sense unless I'm going for scaled production.

terryspitz, 2023-06-21:
Update: I managed to order min. quantity 5 of these from JLCPCB for £28, including the USB socket, but excluding the microswitch, crystal and inductor which I got from LCSC and soldered myself.  They work for programming an on-circuit PFS154 SO8, but don't calibrate successfully (I didn't realise I needed to expose PA4 pin).  

With other parts, shipping & duty (to post-Brexit UK) I'm willing to sell the extras at cost: £12/$15/€14 plus shipping to your location.