free-pdk/easy-pdk-programmer-hardware #22: Nothing found

Skywalkerf34, 2021-09-07:
Hi,

This look very good projet and I wanna have programmer for some cheap board later so 
I have done assembly for 5 Programmer using provided GERBER files
I successfully program the ST32 chip
I am under windows 10 and install STM32 driver
Using easypdkprog list or --version I get answer (confirmed version 1.3)
But when I wanna try the probe command the answer is "probing IC... Nothing found" with a PFS154 or PFS173 fresh new chip

This append on all of the programmers, so I guess something wrong :-5
What checking on PCBA can I run to fix this ?
What command can I pass to check all ?

Rgds

freepdk, 2021-09-17:
You should compile and run easypdkprogtest and then check the output voltages reported from easypdkprogtest + measure the voltages on VDD and VPP.

To compile easypdkprogtest on windows, please have a look at the INSTALL file in the software repository: 
https://github.com/free-pdk/easy-pdk-programmer-software/blob/master/INSTALL

(you need to install MSYS2 and install a compiler (gcc) for MinGW or MSYS2 : you can find multiple guides on the internet like this one: https://www.devdungeon.com/content/install-gcc-compiler-windows-msys2-cc)

Skywalkerf34, 2021-09-25:
Hi freepdk,

Thanks for all those links, I have installed MSYS2 and all suggested package
Now my question is how to compile with that tool?
I run "MSYS MinGW 64-bit" from Start menu.  and get a shell interface waiting for some instruction
 Now how I can call  gcc to build  the .exe

freepdk, 2021-09-27:
@Skywalkerf34 :
VDD = 5.00V / VPP = 5.00V are the values easypdkprogtest sends.

Your values are 100% good. This verified that >98% of the PCB is working correct.

The remaining things on the PCB which might be wrong are just the resistors R18,R19,R20,R21,R22 (which is very very unlikely) or the soldering of the STM32 / the header is not good.
=> This should be easy to verify with a multimeter (check for shorts between the data lines on the header, check for continuity to resistors / resistance to the STM32). 

After that said, it might just be that all your PCBs are working correct and you have a problem with the socket where you place the IC inside. Also make sure to place the IC (PFS154 / PFS173) at the bottom of the socket...

Skywalkerf34, 2021-09-27:
@freepdk 
Thank you so much for your support
I have checked the MCU and resistor to header carefully and everything is right
But still same answer :-0
the "only" thing I didn't check are those adapter PCB design for PFS and PMS do you know if are they right about the design (schematic) ?

freepdk, 2021-09-27:
@Skywalkerf34 
Which adapter PCBs are you refering to? The programmer is designed to use of the shelf sop16 adapter to be plugged in the programmer directly:  google search "sop16 adapter" => https://www.google.com/search?q=sop16+adapter

Skywalkerf34, 2021-09-27:
@freepdk 
those here : https://github.com/free-pdk/easy-pdk-programmer-hardware/tree/master/adapter-pcbs
I think I put the finger on...those are for SOT6  :-5

Skywalkerf34, 2021-09-27:
Better read...10 times
I put the question seeing that PCB was exactly pinout form PADAUK DATASHEET and remembering designing those PCB crossing pin...

freepdk, 2021-09-27:
@Skywalkerf34 
Unfortunately I don't have any experience with the adapter from here:  https://github.com/free-pdk/easy-pdk-programmer-hardware/tree/master/adapter-pcbs
For me it looks like you need the exact same socket as the creator of the adapter PCBs used. In case you just have a similar looking socket the pinout of the socket might be different.

I created a wire based adapter myself in the past... It took me also some tries to get the wiring right. So maybe something is wrong with your adapter.

Skywalkerf34, 2021-09-27:
@freepdk 

My mistake! 
I was stupidly thinking PADAUK did different pinout for the PFS and PMS series, so the adapter was necessary
As I solded my SOP 16 adapter to the universal I just make a "reverse" for bottom to plug on the programmer and... 

TADA.... IT WORKS !!!!!! 

The so call adapter are only necessary for the small SOP6 models but I didn't paid so much attention about  PADAUK datasheet PINOUT nor the "SOP6" indication on the adapter page....I shoud have read more carefully  again and again and again

freepdk, 2021-09-27:
@Skywalkerf34 

> TADA.... IT WORKS !!!!!!

Great. Closing this issue now :-)

closed 2021-09-27 by freepdk

Skywalkerf34, 2021-09-27:
Now I got one working I can fix the few other programmer I have left
Maybe the next step would be for me to re-design a little the PCB with some change (diode really too small; OSCILLATOR could be SMD etc...
But you already do a GREAT JOB !!!!

freepdk, 2021-09-27:
There is already a re-designed PCB (called "lite") which reduced parts (no diodes at all).

The OSC and many other parts been chosen just by "lowest price" on LCSC almost 2 years ago... Back then the complete BOM was <US$2 :-)

Things changed a bit.