free-pdk/easy-pdk-programmer-hardware #23: vpp too low

Skywalkerf34, 2021-09-27:
@freepdk 

I have one of this little series that give VDD 5.00 butVPP 0.65 volts ( screen and measured)
Any suggestion what to check ?

Kind regards

Skywalkerf34, 2021-09-27:
@freepdk 
On the first one I double check all those resistors up to the MCU pin seem ok but not better, 
so I change OP  with same bad result :-[

Skywalkerf34, 2021-09-27:
vref is 3.34

freepdk, 2021-09-27:
Since you have a working device you should be able to check some things quickly:
- start easypdkprogtest on the working device and measure + write down all voltages you see on the opamp pins
- start easypdkprogtest on the problematic device and do the same
=> This will show differences and might lead to the solution.

freepdk, 2021-09-28:
I added a picture with measurements from a good programmer:

[image]

Skywalkerf34, 2021-09-29:
@freepdk 

Again, thanks a lot for your help
I order some new inductor that seem very similar to the reference you give in the BOM as the original manufacturer seem not popular here in my place.
It is supposed to be sent today by express and expected at home in about 4 days
As I receive I will swap the inductor and have to check about all voltages you kindly print for me.
I Will give you news then

freepdk, 2021-09-29:
More suggestions:
- check if L3 soldering is creating a short on PCB
- check orientation of U6 (MT3608), maybe it's rotated 180deg (pin1 of MT3608 is closest to the coil)
- remove L3 and measure voltage on C1 / C4 => should have 0V without L3

Skywalkerf34, 2021-10-04:
@freepdk 

thanks a lot again for your kind support
I receive the new inductor and replace on the 2 "working" PCB
the results are exactly the same, with under voltage between C4 and D1
What the rule of Q1 that is driven by STM32 ? it should "drive the power in for the MT3608  right ?
Then what config I should run to check the voltage at that measuring point? for the moment I just plug the programmer to measure
If I remove R24 I notice that now voltage become almost 15v ( 14.4v exactly)

freepdk, 2021-10-04:
@Skywalkerf34 

You should run the easypdkprog test  which outputs 5V on VDD and VPP. 
The transistor (and the 15V output) should be enabled as soon as you connect the programmer to to USB .

>If I remove R24 I notice that now voltage become almost 15v ( 14.4v exactly) on one of the board
This is an interesting find. This suggests that (maybe after the opamp) something is pulling the output down (e.g. a small short - maybe solder bridge or pcb manufacturing problem, a shorted capacitor, ...).
You can try this theory by removing C2,C10,R15,R17. 
easypdkprog might be unresponsive (since firmware might hang during startup if ADC measurement is missing) but you still should be able to make a measurement.

Skywalkerf34, 2021-10-04:
@freepdk 

Hi,
Hope you are going well,
I try to remove those one by one with no change
removing/putting back R24 confirm this problem on the second board too

Easypdktest was giving quite good value. the problem seems to be on 15v line

This is really strange and at that step of understanding your schematics, I don't understand where can come from...

Skywalkerf34, 2021-10-04:
Should I try to remove  R14 and R20 pull down resistor ?

Skywalkerf34, 2021-10-04:
@freepdk 

Removing R14 and R16 didn't change  :-(

Skywalkerf34, 2021-10-04:
@freepdk 

I check the BOM and my boards and I notice that I use AS358MTR  your BOM said AS358**A**MTR
datasheet just said "_The AS358A has more stringent input offset voltage than the
AS358"_ I am not sure what that mean as there is no other explanation about!

What do you think?
I think there is a mistake somewhere as all the boards are reaction same way, but without deep understanding the schematic it is not so easy to found out what's wrong

freepdk, 2021-10-04:
@Skywalkerf34 

* AS358 and AS358A should make no difference
* When you remove R24 you completely remove the opamp AND ALL FLLOWING components from the DCDC output. So it might be that the opamp just works fine but something else after the opamp pulls down the signal. For instance in case you would solder a resistor instead of a capacitor for C2 or C10 ... or there is a solder bridge at C2 or C10 or .... the output of the opamp would be pulled down which can pull down the complete voltage of the DCDC...
* removing R14/R20 will not help at all. This are part of voltage divider for the ADC (voltage measurement) in STM32. In case you remove them then STM32 will receive to high voltage on ADC pins and this might destroy the STM32.
* => Please remove the components I suggested: C2,C10,R15,R17.

Skywalkerf34, 2021-10-04:
@freepdk 
_=> Please remove the components I suggested: C2,C10,R15,R17._

Already try without any better result :-(

freepdk, 2021-10-04:
Can you take a high resolution picture of your PCB (from the most bad PCB you have) and attach it here?

Skywalkerf34, 2021-10-04:
[image]
[image]
[image]
[image]
[image]

freepdk, 2021-10-04:
@Skywalkerf34 

* **C4 seems much to small** (the bigger 1206 component was chosen for a reason). What cap did you use there (link to datasheet).
* soldering and orientation of ICs is looking ok
* picture 2 and 3 looks like some components (caps near the STM32 / resistors bottom left) might have solder bridges or are shifted to much and might create a short. You should add some flux and then hot air will pull them to correct positions and move the solder to to correct spots.

Skywalkerf34, 2021-10-04:
@freepdk 
Capacitors near STM32 checked,
I cannot find the information about the C4 I used that time so I try to change for one I currently use on my design 10µF tantalum.....no change :-[

Skywalkerf34, 2021-10-04:
shit, I just discover that it was not 4.7µf but 47µf for C4.....

Skywalkerf34, 2021-10-04:
I am using schematic not the BOM list every time, maybe I should change to stronger glasses ;-)

Skywalkerf34, 2021-10-04:
@freepdk 
So, I changed again C4 with 47µF polarized(chemical) capacitor result eep very similar with only 9.2 volts between D1 and C4

freepdk, 2021-10-04:
@Skywalkerf34 

Do you have an oscilloscope ?

Skywalkerf34, 2021-10-04:
@freepdk 
Yes for sure

freepdk, 2021-10-04:
I will do some test measurements later which might be helpful to get a clue what's going on with your boards.

To be honest, when I created and assembled the first programmer it immediately worked. I only had a solder bridge at the STM32 which shorted the two output pins connected to the programming header.
Since all your boards are affected something really strange is going on. Curious to find out :-)

Skywalkerf34, 2021-10-04:
@freepdk 

This append to me time to time doing  new prototype design....It works immediately and confirm the design... but going to pre-serie then problem occurs ;-)

I do electronics dev for many years but now mainly from 80x51 to microchip then ATMEL but now almost everything with ESP32. 
I have my own company and small assembly line for SMT.
Now I was looking for cheap MCU for very light (or cheap) project.

I really appreciate your kind help to solve the problem

freepdk, 2021-10-10:
@Skywalkerf34 

The labeling suggest that this is a "LM358" from Texas Instruments. 
However all genuine Texas Instruments LM358 ICs I have seen always have a good visible dot for pin 1. Also the Texas Instruments logo looks a bit strange as well. 
In case this is a relabeled (cheap) opamp, chances are high that it will not work well / will not work at all with voltages >5V.

Skywalkerf34, 2021-10-11:
@freepdk

Thanks again for your support
I am convience there is global problem somewhere, and we are pointing the OP level part for several time.
I was out those days, Will will try to measure the output of the step-up but what was about clear is that without R24 wich power the OP the voltage was correct. 
On same way I put back the resistor and remove OP and get same result (right voltage)

I will put order for some fresh new OP to an other shop and see what append then

Will keep you informed as I get some result

Skywalkerf34, 2021-10-30:
@freepdk 

Hi,

I receive the new OP and change on one of the most "working" board but no change :-[
So I come back to one I put on the side that was given already a bad vpp value and discover that that board give good 15 volts and except vvp other voltage you kindly indicate was about ok. So I check that last one an just press a little more pin18 of STM32...tada vpp come good then :-)
So I set a pfs154 on the adapter and probe => work !
erase => work !
program helloword => work !
 start => hello world!   

So, by the end, This is just my original wiring of the board that work good now with the LM358 and that "small capacitor  on C4 and some inductor I bought at the beginning.

Now, thanks to your help, I have at least ONE programmer that works and I can now investigate why the other didn't run well up to now.

Thanks again for all your patience and great help to guide me on the way to fix

Skywalkerf34, 2021-11-05:
@freepdk
I am still facing problem on the 4 other Programmer can you be kind to explain the purpose of Q2 and Q1 and confirm D2 and D3 are connected to Q2 ( didn't find information on the schematic)